airpartners/logger #39: Review Draft Sensor Box Documentation

anushadatar, 2021-03-18:
I've made a draft of more extensive sensor box documentation; I did not invent it (in fact, shoutout evan for all the work on it) but I did add a bunch to it to make onboarding and potential handoff easier. That being said, it's still in draft form and I would really appreciate feedback and some more sets of eyes on it. Please take a look at the wiki - if it's something small or something you feel strongly about, go ahead and directly edit it, and if it's something more complicated/something you want me to do, feel free to get in touch with me or (ideally) leave a comment on this issue. Also let me know if it looks good to you (or if you don't want to look at this) so I can close this issue once people are happy.

hsharriman, 2021-03-18:
I got to take a quick look at it, and wow this is so thorough and lovely!!! I think it looks good to me.. I am pretty clueless when it comes to hardware and I think I could follow along and set up a sensor box by referencing the wiki.

davidt315, 2021-03-18:
My notes:
Deployment page looks great!
Hardware overview is great, added a month for current state
Next steps is awesome
Provisioning and runbook seem to be super thorough and great
Software overview and troubleshooting also look great

this probably took a long time lol big high five to you!

Inkering, 2021-03-18:
I changed one tiny typo on the motor controller section. Thanks for this! it's so thorough : D

Structurally maybe the next steps should get an update now // in a little bot? Or an update on things from December.

anushadatar, 2021-03-18:
> I changed one tiny typo on the motor controller section. Thanks for this! it's so thorough : D
> 
> Structurally maybe the next steps should get an update now // in a little bit? Or an update on things from December.

Updated the next steps for now at least: https://github.com/airpartners/logger/wiki/next-steps. I'll probably lean on GH issues quite a bit at least for now those as those are easier for me to keep track of, but you're right that we are at enough of an epoch that we should make an update.

anushadatar, 2021-03-22:
@wFairmanOlin Let me know at your convenience so I can make whatever changes you'd like before the semester gets more chaotic with other work :)

wFairmanOlin, 2021-03-22:
Looks super good! Great starting point for anybody picking up this project later.

anushadatar, 2021-03-22:
Closing this for now, but feel free to make changes or request changes as you would like or need,

closed 2021-03-22 by anushadatar